bigtreetech/MMB #1: Sensorless not working over Canbus and USB

thiagolocatelli, 2023-09-26:
Trying to run sensorless and keep getting this error:

```
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/mcu.py", line 74, in _do_send
    return xh.get_response(cmds, self._cmd_queue, minclock, reqclock)
  File "/home/pi/klipper/klippy/mcu.py", line 50, in get_response
    raise serialhdl.error("Timeout on wait for '%s' response"
serialhdl.error: Timeout on wait for 'tmcuart_response' response

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/homing.py", line 235, in manual_home
    hmove.homing_move(pos, speed, triggered=triggered,
  File "/home/pi/klipper/klippy/extras/homing.py", line 71, in homing_move
    self.printer.send_event("homing:homing_move_begin", self)
  File "/home/pi/klipper/klippy/klippy.py", line 263, in send_event
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/klippy.py", line 263, in <listcomp>
    return [cb(*params) for cb in self.event_handlers.get(event, [])]
  File "/home/pi/klipper/klippy/extras/tmc.py", line 524, in handle_homing_move_begin
    self.mcu_tmc.set_register("TCOOLTHRS", tc_val)
  File "/home/pi/klipper/klippy/extras/tmc_uart.py", line 249, in set_register
    self.ifcnt = self._do_get_register("IFCNT")
  File "/home/pi/klipper/klippy/extras/tmc_uart.py", line 230, in _do_get_register
    val = self.mcu_uart.reg_read(self.instance_id, self.addr, reg)
  File "/home/pi/klipper/klippy/extras/tmc_uart.py", line 177, in reg_read
    params = self.tmcuart_send_cmd.send([self.oid, msg, 10])
  File "/home/pi/klipper/klippy/mcu.py", line 78, in send
    return self._do_send([self._cmd.encode(data)], minclock, reqclock)
  File "/home/pi/klipper/klippy/mcu.py", line 76, in _do_send
    raise self._error(str(e))
gcode.CommandError: Timeout on wait for 'tmcuart_response' response
```

Configuration

```
[duplicate_pin_override]
pins: kms_mcu_1:PB2,kms_mcu_1:PC15

[tmc2209 manual_extruder_stepper kms_mmu_feeder_T0]
uart_pin: kms_mcu_1:PA10
interpolate: True
run_current: 0.90
#hold_current: 0.1
sense_resistor: 0.110
#stealthchop_threshold: 999999
diag_pin: ^kms_mcu_1:PA3
driver_SGTHRS: 75 

[manual_extruder_stepper kms_mmu_feeder_T0]
step_pin: kms_mcu_1:PB15
dir_pin: !kms_mcu_1:PB14
enable_pin: !kms_mcu_1:PA8
rotation_distance: 22.6789511	
microsteps: 16  
full_steps_per_rotation: 200 #200 for 1.8 degree, 400 for 0.9 degree
velocity: 35
accel: 150
endstop_pin: tmc2209_kms_mmu_feeder_T0:virtual_endstop
endstop_name: kms_sensorless_endstop_1
# Uncomment two lines below to give option of  selector "touch" movement and homing
#extra_endstop_pins: tmc2209_kms_mmu_feeder_T0:virtual_endstop,kms_mcu_1:PB2
#extra_endstop_names: kms_virtual_endstop_T0,kms_splitter_endstop_1
```

Collapsednut, 2023-10-06:
Just a random thought cause I'm also going to be using sensorless on this board. I assume you have a jumper on the diag pins for driver 1 and I see that is linked to the PA3 pin of stop-1 on the schematic. Maybe try manually with a jumper wire going to a different stop pin like PA4 or PB8 and define in config. Maybe just maybe the trace on the board connecting diag1 to PA3 is faulty is what I'm getting at.

thiagolocatelli, 2023-10-06:
@Collapsednut I have tried all that and also checked for continuity between all diag pins

ntchris, 2023-11-10:
for ERCF / happy hare v23 ... I think sensorless is for gate selector , not for extruder ? 

Also , hope this can help a little bit maybe
https://github.com/ntchris/voron_mod_super_diy/tree/main/for_enraged_rabbit/btt_ERCF_Canbus