bigtreetech/MMB #5: Documentation needs to be improved around function of CAN vs. USB vs. HVIN

kieraneglin, 2023-11-27:
It is very unclear that you MUST use the CAN connector to power the board and cannot use the HVIN + USB connectors alone.

For future readers, here's how the various connectors work for real-world usage:

- If you're flashing firmware, you can connect just USB with the `VUSB` jumper in place. >>> _Be certain to remove the `VUSB` jumper before connecting any power or CAN cables_ <<<
- If using CAN as a communication interface and you don't need extra juice for the steppers, just connect the CAN plug
   - Set each stepper voltage jumper to `VIN`
   - If your mainboard supports CAN but you otherwise don't know what to do, this is probably the option you want
- If using CAN as a communication interface and you DO need extra power for the steppers, connect the CAN plug and also connect your additional power leads to the `HVIN` and `GND`
   - Set each stepper voltage jumper to `HVIN`
-  If using USB as a communication interface and you don't need extra juice for the steppers, you have to connect the CAN plug and put 12-24v through its power leads. You then have to connect a USB cable from your printer to the MMB
  - Set each stepper voltage jumper to `VIN`
    - If your mainboard DOES NOT support CAN but you otherwise don't know what to do, this is probably the option you want
-  If using USB as a communication interface and you DO need extra power for the steppers, you have to connect the CAN plug and put 12-24v through its power leads and also connect your additional power leads to the `HVIN` and `GND`. You then have to connect a USB cable from your printer to the MMB
   - Set each stepper voltage jumper to `HVIN`

tmshader, 2024-07-24:
Why can't HVIN work with the VUSB jumper? I tried it with the v1.1 board and it works flawlessly. I have my stepper jumpers on HVIN and I power the rest of the board with the USB cable and I haven't had any problems yet.

AustinTylerDean, 2024-08-12:
> Why can't HVIN work with the VUSB jumper? I tried it with the v1.1 board and it works flawlessly. I have my stepper jumpers on HVIN and I power the rest of the board with the USB cable and I haven't had any problems yet.

I really hope this is correct, and that BTT comes back and says "Yes, do this." 

It's terribly confusing when you connect the HVIN 24V and GND, pull the jumper, connect USB, and turn the thing on only to not see anything light up! Thanks to OP for the info as well!

AustinTylerDean, 2024-08-12:
> Why can't HVIN work with the VUSB jumper? I tried it with the v1.1 board and it works flawlessly. I have my stepper jumpers on HVIN and I power the rest of the board with the USB cable and I haven't had any problems yet.

I am not an electrical engineer, but I did take a stab at the schematic. I can't tell if the servo is getting power from the USB or the HVIN. I started getting confused. My worry is that these servos are expecting 4.8-6v. I don't know if VUSB is getting bumped to that or not. Would love some confirmation. I have sent an email to BTT. We will see what that brings.

tmshader, 2024-10-15:
So I had time to review the schematics and as far as I can tell, the USB connects without any voltage conversion to the same +5V rail that the DC_IN of the CANBUS connector connects to, but the CANBUS is stepped down to 5V, so they are both used for powering the servo. The HVIN is only connected to the motor drivers and only if those specific jumpers are selected. So it should be safe to use CANBUS to power and USB to communicate, but you **HAVE** to power the board through CANBUS because the USB connection cannot provide enough amps to drive the servo. And you can use HVIN with a secondary power supply to give power to the motors, if your CANBUS can't provide enough amps to drive the motors and the servo simultaneously.

pahofmann, 2025-04-30:
Glad I found this, sad it hasn't made it to the manual after years...