bigtreetech/MMB #7: Wrong Pins

oMTEo, 2023-12-02:
Im on build a ERCF with your MMB-Board and found 2 wrong declared pins, Ive fixed it in the following pic (red arrows)

[image]

Xonman, 2024-01-12:
@Motalw hopefully you have got it working by now, but the manual incorrectly shows ABAB. Steppers should be wired AABB to work correctly.

colindembovsky, 2024-04-07:
@Xonman @Motalw I had the same issue and wiring AABB worked for me.

kullwarrior, 2024-05-10:
V1.1 Pins slightly changed.
[image]

mrellifant, 2024-05-18:
Hi @kullwarrior - is that diagram correct for the v1.1? the stepper 1A 1B 2A 2B doesn't match the manual which was also updated for v1.1 last week (about the same time as your post) its 1B 2B 1A 1B - and this is the same as the ERCF v2 manual.
I'm having troubles getting NEMA17 steppers to work properly using the latter (and also finding the proper pinouts for the trianglelabs TL-42BYGH20 which they're suppling in their ercf v2 kit). 
thanks Wayne.

dbookusa1, 2024-06-15:
That is the difference between a v1.1 and the V1.0.