bigtreetech/MMB #11: CAN bus wiring with EBB SB2209, U2C, and MMB

ksteddom, 2024-06-02:
What is the preferred way to wire the CAN bus when you have multiple devices? In my example I have the SB2209 tool head board connected to my pi via a U2C adapter. What order should the three devices be in the bus and which ones get the 120R resistor jumpers?

alidaf, 2024-06-24:
120R jumpers go on the start and end points of the CanBUS so on your U2C adapter and SB2209 toolhead board. Your Pi isn't part of the CanBUS - it's just communicating over USB with the U2C, which implements CanBUS.

alidaf, 2024-06-24:
I think in your case all signals originate from the U2C so it is the start. If you have 2 cables from the U2C, 1 to the the MMB and 1 to the toolhead then you have a star network, the peripherals are still end devices and both need jumpers along with the U2C. If you have 1 cable from the U2C to the toolhead, and then 1 from the toolhead to the MMB then these would be daisy-chained so only the U2C and MMB need jumpers (the toolhead is in the middle). I've never seen this config in practice as the toolhead would need another CanBUS socket and I've not seen any like this.