bigtreetech/MMB #16: MMB v2 missing files

BlackNet, 2025-01-27:
MMB v2 is missing the hardware layout/ pinouts and step file.

looxonline, 2025-01-27:
At the moment the pins information is contained within the user manual. The STEP file will be added after the Spring festival ends. @BTTJUSTIN

BlackNet, 2025-01-30:
Have no clue what you are on about with spring. Spring of which year?  

Why so long and what's the delay about?

randybb, 2025-01-30:
Because no one works during Chinese New Year (= Spring Festival).

Manu512, 2025-02-16:
Missing files and Wrong parameters.
2 days lost because documentation say PD0 and PD1 for can.
With PB0 and PB1 all works.

BlackNet, 2025-02-16:
What we ended up doing was take an ohm meter and probe the chip and found the right pin. [image]

BlackNet, 2025-02-24:
> As I hold in my hands a version 2 board... and since chineese New Year break is well over now...
> 
> Can you please post missing schematics, and pin definitions particularly for MCU_STP5 thru MCU_STP16 as they are located on a 14-pin connectors not clearly labeled. Please also specify the connector required as it would help newbies...
> 
> Also, please indicate MCU_SWD header pinout.
> 
> Lastly, please clarify can bus pin (on PD0/PD1) or (on PB0/PB1) ???
> 
> Thanks so much in advance, as it woill enable using your newly developed hardware!!!

I can confirm I have a functional MMB2 board using CAN, and it does indeed use the pinout photo I listed earlier on PD1.

`
CONFIG_STM32_MMENU_CANBUS_PD0_PD1=y
CONFIG_STM32_CANBUS_PD0_PD1=y
`

I do not have listings for stp5-16.

bigtreetech, 2025-02-25:
> Missing files and Wrong parameters. 2 days lost because documentation say PD0 and PD1 for can. With PB0 and PB1 all works.

Version 2 is most certainly PD1 and PD0. There is no possible way that PB would work for CAN. All of the required information is in the manual. STEP files have been uploaded now.

closed 2025-02-25 by bigtreetech

m3cinc, 2025-02-28:
Can you please post the stop pin layout in more details. It is not very precise to just mention stp5-16 without a pin diagram.
Thanks in advance.

looxonline, 2025-02-28:
All of the pins for stop pins have their pin numbers labeled on the existing pin diagram.

m3cinc, 2025-02-28:
There are 2 rows of pins indeed labeled PC7, PA9, etc... 
My question is which stop pin do they represent? a corresponding table between stop # and pin # is needed.
Thanks in advance.

looxonline, 2025-02-28:
Does it matter though? Ultimately it is just a stop pin and you can call each one whatever you want in the firmware. You assign it a function in the cfg file.

m3cinc, 2025-02-28:
Agree, but when you configure and build multiple units, you like to have
all the specifications identical... I am using klippain to program and
would hate to have multiple configurations for these units.