bigtreetech/MMB #19: Flash Firmware through USB ends im [src/stm32/Makefile:111: flash] Error 255

LordMM74, 2025-02-22:
I tried do flash firmware through usb like described in docs.

`pi@RatOS:~ $ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

`pi@RatOS:~/klipper $ make flash FLASH_DEVICE=0483:df11
  Flashing out/klipper.bin to 0483:df11
sudo dfu-util -d ,0483:df11 -R -a 0 -s 0x8000000:leave -D out/klipper.bin

[sudo] password for pi:
dfu-util 0.11-dev

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
Opening DFU capable USB device...
Device ID 0483:df11
Device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 011a
Device returned transfer size 1024
DfuSe interface name: "Internal Flash   "
Downloading element to address = 0x08000000, size = 30928
Erase           [=========================] 100%        30928 bytes
Erase    done.
Download        [=========================] 100%        30928 bytes
Download done.
File downloaded successfully
Submitting leave request...
dfu-util: Error during download get_status: -9 (LIBUSB_ERROR_PIPE)

Failed to flash to 0483:df11: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=0483:df11

make: *** [src/stm32/Makefile:111: flash] Error 255
`

I retried step by step the docs several times.

The BTT MMB CAN is connected through usb to raspberry pi.
The VUSB jumper is set to power the board through usb. 
Botting into DFU mode seems to work ... lsusb shows the board.

closed 2025-02-22 by LordMM74

8bit-boom, 2025-03-06:
I have same problem, did you find solution to it?

bigtreetech, 2025-03-07:
Ignore this error if it happens. What matters is that the firmward download reaches 100%